tobozo/ESP32-USB-Soft-Host #3: invalid conversion from 'int' to 'timer_autoreload_t'

ESP32DE, 2021-03-19:
small conversion problem

instead
config.auto_reload = 1
better
config.auto_reload = (`timer_autoreload_t`) 1;

suggestion: https://github.com/tobozo/ESP32-USB-Soft-Host/pull/2

tobozo, 2021-03-19:
hey Rudi, thanks for your feedback, I'll ship this fix in the next update. 

Can you specify the environment you got this error from ? 

Arduino IDE doesn't seem to complain about the lack of that type, is that invalid conversion error from vscode, eclipse, platformio ...?

tobozo, 2021-03-19:
your fix is in the code although I still have no idea where it happened, I'll close this issue and your PR too, thanks again for your feedback !

https://github.com/tobozo/ESP32-USB-Soft-Host/blob/main/src/ESP32-USBSoftHost.hpp#L103

closed 2021-03-19 by tobozo