tobozo/ESP32-USB-Soft-Host #4: Connection issue

zbx-sadman, 2021-04-09:
Hello.

I want to ask for schematic.  

Do you put +5V on USB connector, and connect D+/D- directly to ESP32 pis (without voltage level shifter)? 

I have tried connect some devices to ESP32 thru cheap 5V<->3.3V level shifter and have no success. Keyboard and one mouse with long cable (~1.4m) is not recognized, and another mouse with shorter cable (~0.8m) is unstable recognized .

Unfortunately i'm not sure about ESP32's pins 5V toolerance and i affraid to burn out microcontroller.

Thanks for your time.

tobozo, 2021-04-09:
hey @zbx-sadman thanks for your feedback.

I don't use a level shifter but I haven't managed to get every mouse detected, looks like some models are using high speed usb for some reason.

Is your other mouse being unstable even when the keyboard isn't plugged in? If so you may try to power the USB separately (but keep the GND common).

ESP32 5v tolerance is a bliss, and burning and ESP32 is an achievement :-)

zbx-sadman, 2021-04-09:
I tried other ( https://robotdyn.com/logic-level-converter-bi-direction-4channel.html ) level shifter based on BSS138  and have no success too with all my mouses/keyboards
Then i maked direct connection between ESP32 and short-cabled mouse (M-UV55a) with all wires (5V/GND/D+/D-), and seen miracle:

[image]

Then i tested long-cabled Logitech B100 mouse, and Logitech K120 keyboard. All parts showed activity in the Serial Monitor.  I guess that's is win.

May be later i'll return to this quest with  oscilloscope.

Many thanks @tobozo and @sdima1357

closed 2021-04-09 by zbx-sadman

zbx-sadman, 2021-05-09:
Some good news.

I've tested voltage level on the DP/DM level's and seems it good for the ESP32 inputs. I've seen 3.7v max on peak-peak metric ("Pk-pk" on oscilloscope screen).

[image]