tobozo/ESP32-USB-Soft-Host #6: scan codes

nogueira, 2021-05-05:
This lib is perfect, I just don´t know how to decode the data received to scan code.

Can you help me?

tobozo, 2021-05-05:
Hey Octavio, thanks for the feedback,

The sketches in the example folder implement the data decoding for a keyboard only (unfortunately I couldn't get a mouse to work).

The `KeyboardReportParser` used in the examples is heavily based on [USB_Host_Shield_2.0](https://github.com/felis/USB_Host_Shield_2.0) source code, so if you're trying to detect something else it may be a good start.

zbx-sadman, 2021-05-07:
@nogueira , it simple for standart HID keyboards with standart "keyboard report" (as my  Logitech K120).

Format is: 1 byte for key modifiers (Alt, Ctrl, Shift, Win) - 1 byte reserved (always 0x00) - 6 bytes for up to six keys pressed simultaneously.
 
HID codes you can find here: ["HID Usage Tables FOR Universal Serial Bus (USB) Version 1.21"](https://www.usb.org/sites/default/files/hut1_21_0.pdf) (Keyboard/Keypad Page (0x07) ) 

 
Key modifiers are defined in the IEEE HID Spec as follows:
```
LEFT  CONTROL          1    # 00000001 as binary
LEFT  SHIFT            2    # 00000010
LEFT  ALT              4    # 00000100
LEFT  CMD|WIN          8    # 00001000
RIGHT CONTROL          16   # 00010000
RIGHT SHIFT            32   # 00100000
RIGHT ALT              64   # 01000000
RIGHT CMD|WIN          128  # 10000000
```
For example:

```
Right arrow (0x4f) pressed
in :00 00 4f 00 00 00 00 00 
All keys released
in :00 00 00 00 00 00 00 00 

Left arrow (0x50) pressed
in :00 00 50 00 00 00 00 00 
All keys released
in :00 00 00 00 00 00 00 00 

Enter (0x28) pressed
in :00 00 28 00 00 00 00 00 
All keys released
in :00 00 00 00 00 00 00 00 

Ctrl (0x10, B0001 0000) pressed
in :10 00 00 00 00 00 00 00 
Ctrl+Enter (0x28) pressed
in :10 00 28 00 00 00 00 00 
Enter (0x28) released
in :10 00 00 00 00 00 00 00 
All keys released
in :00 00 00 00 00 00 00 00 
```