tobozo/ESP32-USB-Soft-Host #7: Not an Issue: Possible to add support to send data?

cpriest-gc, 2021-07-09:
Hi,

I appreciate this isn't an issue, but just wondering if it's possible to add support to send data?

I have a universal HID device I created that I'd like to talk to via USB Soft Host, would it be difficult to implement?

tobozo, 2021-07-09:
Hi, thanks for your feedback

it's probably worth asking that question on the [upstream repository](https://github.com/sdima1357/esp32_usb_soft_host), if it is indeed possible then the Arduino library wrapper will inherit from it.

Are you trying to trigger the rgb leds of a mouse/keyboard ?

cpriest-gc, 2021-07-10:
Hi Tibozo,

No, not a keyboard, a HID device that I created that's a USB slave and uses an atmel mega processor with a USB transceiver chip that acts as a USB slave, but is a HID universal device, just need to send 4 bytes to it at a time :)