tobozo/ESP32-USB-Soft-Host #10: Does not support digispark, USBHID emulation

mrguai, 2021-08-29:
Does not support digispark, USBHID emulation

mrguai, 2021-08-29:
digispark is a USB input device

tobozo, 2021-08-29:
Hi, thanks for your feedback,

do you mean Digispark BadUSB ?

I don't have one so I can't test but my guess is it's not a USB-LE device as this library requires/

mrguai, 2021-08-29:
yes,Digispark BadUSB
My test seems to be unable to carry out USB enumeration, but the USB package cannot be caught

tobozo, 2021-08-29:
maybe adding a `delay(5000)` in the `setup()` will let the esp32 some time to boot and start listening before the atTiny starts USB emulation ?

mrguai, 2021-08-29:
I try, how do I feel that USB input devices are not supported。
Because, I wrote two programs to compare.
1. The led flashes, just plug it in and it will flash
2. The led flashes + keyboard, the led code is behind the keyboard. Plug in the USB, the led does not flicker

tobozo, 2021-08-29:
I assume you're basing your research on this [wiki](http://digistump.com/wiki/digispark/tricks).

it says: 

> When the digispark boots up, it uses the USB signal to calibrate its clock to 16.5mhz

Now I'm thinking maybe this can't work with usb-soft-host, or maybe it needs a careful calibration with an oscilloscope, but I can only speculate.

mrguai, 2021-08-29:
ok thank

nogueira, 2021-08-30:
Any chance to make it work with a barcode reader? It emulates a keyboard but the esp32 don´t recognise it

tobozo, 2021-08-30:
@nogueira only if the barcode reader emulates a USB-LS device

closed 2025-05-08 by tobozo