tobozo/ESP32-USB-Soft-Host #12: Is it possible to translate the device to another?

harshilishere, 2021-09-24:
to keep it simple, let's say i have a mouse but i won't to fool my computer into thinking it as a keyboard(sorry for a lame example), is that possible??

The real world application can be, sometimes we don't have the exact device on the place to test. We can get the device's usb info, duplicate it and use or the test the software temporarily!!

Is this possible somehow?!

tobozo, 2021-09-24:
> Is this possible somehow?!

A device descriptor cloner is very unlikely with this library as it is limited to USB-LS.

It sounds like you're looking for an ESP32-S2 or S3.

closed 2021-09-24 by tobozo

harshilishere, 2021-09-24:
Sorry i have no idea about ESP32-S2 or S3. Can you please tell me what is it?

tobozo, 2021-09-24:
same as ESP32 but with built-in hardware USB (instead of software USB limited to Low Speed devices)

https://www.espressif.com/en/products/socs/esp32-s2

harshilishere, 2021-09-25:
@tobozo ah understood! thanks! you may close this now.