tobozo/ESP32-USB-Soft-Host #13: Can't compile "fatal error: hal/cpu_hal.h: No such file or directory"

Fredkopp, 2021-09-29:
Feeling kinda dumb, but I'm getting this error trying to compile the usb_test.ino. There is no such file in this library, but the "usb_host.c" is referencing it. What am I missing? This is the exact error message: 
"/home/user/Arduino/libraries/ESP32-USB-Soft-Host-main/src/usb_host.c:18:25: fatal error: hal/cpu_hal.h: No such file or directory
compilation terminated."

tobozo, 2021-09-29:
hey thanks for your feedback :+1: 

looks like you're using an older version of arduino core for esp32, the requirement for this driver is [arduino-esp32 2.0.0](https://github.com/espressif/arduino-esp32/releases/tag/2.0.0) and should be available in your boards manager.

Fredkopp, 2021-09-29:
Hey that did the trick! Thank you very much!

tobozo, 2021-09-29:
closing this as solved, have fun !

closed 2021-09-29 by tobozo

Vineeth-B007, 2022-10-07:
In arduino board manager the latest version is not available for installation,still the latest available for updatation is ESP32 1.0.6,
Also tried downloading from github still also doesnt work.
C

tobozo, 2022-10-07:
1.0.6 is quite olde, you probably need to update the package url in the additional boards manager (arduino preferences)

you can remove the old URL and use this one instead:

https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json

the highest version of this package is currently 2.0.5

Vineeth-B007, 2022-10-07:
Yeah its done Sir :)
Thankyou so much.
Can I ask you a doubt,
In my circuit ,ESP32  is flashed via usb through CH340 ,is that possible to make a HID without removing the CH340 as my device is a 2/3 inch thermal printer there is the need of a COM port as well as a HID.

tobozo, 2022-10-07:
this library uses regular pins from the ESP32, so you can do whatever you want with the unassigned pins

Vineeth-B007, 2022-10-07:
Okay sir,
Mine has Zero pins available. Thinking to remove CH340 and use HID ONLY