tobozo/ESP32-USB-Soft-Host #17: Trying to connect a QuickScan qd2430

silentrain23, 2021-12-19:
The Lib works fine with an apple mouse (desc.idVendor = 0x05acdesc.idProduct = 0x0304). 
I need to connect a USB QuickScan. Any idea? Any other usb scanner?
Thankyou

tobozo, 2021-12-19:
hey @silentrain23 thanks for the mouse feedback, I'll add it to the compatibility list.

I have no idea about this barcode scanner model, if it's more than 10 years old there's a chance it's using USB-LS and could be detected as a keyboard, maybe measuring the scanner with an oscilloscope could tell that ?

silentrain23, 2021-12-19:
[@tobozo thankyo](https://www.bsr.at/mediafiles/Handbuch/Datalogic/Quick_Reference_Guide_QuickScan_QD2430_EN.pdf)
it has different settings, but none works fine for me. I have no oscilloscope to make the tests :( The model il from 2015

tobozo, 2021-12-19:
Given the amount of devices listed in the settings it seems unlikely USB-LS is supported anyway.

silentrain23, 2021-12-19:
Do you think it might work if I find a USB-LS in the market?

tobozo, 2021-12-20:
probably if "the market" == ebay 

unless you need bluetooth, an ESP32-S2 can connect to any USB devices, otherwise there's the esp32-s3

silentrain23, 2021-12-20:
Thank you @tobozo but I am working on a OLIMEX ESP32 EVB and I am still waiting to switch to s3. Do you think I can manage to connect a scanner like QUICKSCAN qd2430 directly to ESP32 EVB just using a SoftwareSerial.h and reading the D+ and D- of the USB as TTL? Or is there a better trick around? I am getting crazy :)

tobozo, 2021-12-20:
I'm afraid I can't tell without the product in hands.

Another solution is to use the mini USB Host Shield 2.0 from MAX, a bit expensive but works fine with an [ESP32-Wroom](https://github.com/tobozo/ESP32-3DPrinter-Bridge)

[image]

silentrain23, 2021-12-20:
Thank you, 25€ it's investment I can afford!

closed 2023-04-04 by tobozo