tobozo/ESP32-USB-Soft-Host #18: Won't compile

code32123, 2022-01-05:
I couldn't get the example code to compile without removing all the "inline" tags (on the function declarations). Do these serve a purpose that will cause problems later down the road, or are they safe to remove?

tobozo, 2022-01-05:
hey thanks for your feedback :+1 

Are you using platformio or Arduino IDE, if so what version of the espressif core are you using ?

I'm not even sure what the `inline` keyword is doing inside a C file (that's probably a leftover from the parent project) but if it works without it then it's safe to remove :-)

I'll make some tests soon and update the library acordingly.

tobozo, 2022-01-05:
update: I haven't been able to reproduce the compilation error but I can confirm the library works fine without the `inline` keyword.

code32123, 2022-01-05:
Ok, Thanks! I'm using the Arduino IDE (And I'm not at my computer right now so I'm not sure of the espressif core, but it's an M5Stack Core grey), and it seems to have fixed itself? After removing all the inline tags, it compiled. Once it compiled, I was able to add the inline tags back and it ran perfectly? I restarted the IDE between attempts, so that might have been what fixed it. I am having a different problem that prompted me to attempt this, but I think it's related to power consumption rather than the code. Thank you!

closed 2022-01-10 by code32123