tobozo/ESP32-USB-Soft-Host #20: Don't work USB_Test program

yamachan-hacomono, 2022-07-05:
Hello,

We're testing with a keyboard connected to the ESP32-DevKit C-32E.
But when I type on the keyboard, I don't see the keys.
GPIO has DP connected to 16 and DM connected to 17, with the other pins disabled.
Do you have any advice?

[Env]
ESP32-DevKit C-32E
Arduino IDE 1.8.19 for windows

[Log]
```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4
USB Soft Host Test for Default Wroom
USB#0 (pins 16 17) is OK!
READ_BOTH_PINS = 0100
READ_BOTH_PINS = 0000
READ_BOTH_PINS = 0000
READ_BOTH_PINS = 0000
cpu freq = 160 MHz
TIME_MULT = 77 
D=   0 4800 bits in 1885.000000 uSec 2.546419 MHz  6 ticks in 2.356250 uS
D= 127 4800 bits in 5720.000000 uSec 0.839161 MHz  6 ticks in 7.150000 uS
D=  63 4800 bits in 3775.000000 uSec 1.271523 MHz  6 ticks in 4.718750 uS
D=  31 4800 bits in 2802.000000 uSec 1.713062 MHz  6 ticks in 3.502500 uS
D=  47 4800 bits in 3294.000000 uSec 1.457195 MHz  6 ticks in 4.117500 uS
D=  39 4800 bits in 3042.000000 uSec 1.577909 MHz  6 ticks in 3.802500 uS
D=  43 4800 bits in 3174.000000 uSec 1.512287 MHz  6 ticks in 3.967500 uS
D=  45 4800 bits in 3223.000000 uSec 1.489296 MHz  6 ticks in 4.028750 uS
D=  44 4800 bits in 3204.000000 uSec 1.498127 MHz  6 ticks in 4.005000 uS
D=  43 4800 bits in 3168.000000 uSec 1.515152 MHz  6 ticks in 3.960000 uS
TRANSMIT_TIME_DELAY = 46 time = 4.005000 error = 0.125003% 
USB#1 is disabled by user configuration
USB#2 is disabled by user configuration
USB#3 is disabled by user configuration
```

Thanks.

yamachan-hacomono, 2022-07-15:
@tobozo 
thank you for your reply!

VCC / GND is obtained from a regulated power supply and has a sufficient supply amount.
The connected device is a general USB keyboard. Maybe it's doing fast processing and you don't recognize it. The keyboard is Realforce 91UBK-S.

danieltwagner, 2024-10-15:
Apologies for reviving an old thread but I have the same issue; I'm building against `espressif32@5.2.0` as that's the last version using the tested ESP-IDF 4.4.2 under the hood. 
I'm seeing no callbacks (read: serial output) when connecting or disconnecting a low-speed USB keyboard or typing on that keyboard. The keyboard is a Cherry Stream, and I confirmed syslog shows low-speed.

I tried various GPIO port combinations to no avail. Full serial output below:
```USB Soft Host Test for Default Wroom
TIMER_BASE_CLK: 80000000, TIMER_DIVIDER:2, TIMER_SCALE: 40000000
Attaching onDetectCB
Attaching onDataCB
Attaching onUSBMessageDecode
Attaching onTickCB
Creating message queue
Setting USB Delay
Setting up pins
USB#0 (pins 16 17) is OK!
READ_BOTH_PINS = 0000
READ_BOTH_PINS = 0000
READ_BOTH_PINS = 0000
READ_BOTH_PINS = 0000
cpu freq = 240 MHz
TIME_MULT = 51 
D=   0 4800 bits in 1290.000000 uSec 3.720930 MHz  6 ticks in 1.612500 uS
D= 127 4800 bits in 3806.000000 uSec 1.261167 MHz  6 ticks in 4.757500 uS
D=  63 4800 bits in 2512.000000 uSec 1.910828 MHz  6 ticks in 3.140000 uS
D=  95 4800 bits in 3159.000000 uSec 1.519468 MHz  6 ticks in 3.948750 uS
D= 111 4800 bits in 3480.000000 uSec 1.379310 MHz  6 ticks in 4.350000 uS
D= 103 4800 bits in 3327.000000 uSec 1.442741 MHz  6 ticks in 4.158750 uS
D=  99 4800 bits in 3239.000000 uSec 1.481939 MHz  6 ticks in 4.048750 uS
D=  97 4800 bits in 3199.000000 uSec 1.500469 MHz  6 ticks in 3.998750 uS
D=  98 4800 bits in 3219.000000 uSec 1.491146 MHz  6 ticks in 4.023750 uS
D=  97 4800 bits in 3208.000000 uSec 1.496259 MHz  6 ticks in 4.010000 uS
TRANSMIT_TIME_DELAY = 99 time = 3.998750 error = -0.031251% 
USB#1 is disabled by user configuration
USB#2 is disabled by user configuration
USB#3 is disabled by user configuration
Seleting SCL (blink) Pin #22
Creating timer config
Init timer
Enable interrupt
Register ISR
Start timer
Attaching Timer Task
USB Soft Host Group timer task is now running on core #1 with priority 5```

tobozo, 2024-10-15:
@danieltwagner looks like the keyboard isn't detected 

what keyboard model is this? usb information on that device can be collected from a computer using lsusb or similar utility, it will tell if that device supports USB 1.0 or not.

moreover, you can use `build_flags = -DCORE_DEBUG_LEVEL=5` in platformio.ini to get more information in the serial console, it should print some information when the keyboard is plugged/unplugged

danieltwagner, 2024-10-15:
And:
```
$ lsusb -t | grep "Dev 023"
        |__ Port 002: Dev 023, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 002: Dev 023, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
```
which is in line with the syslog `low-speed` message

tobozo, 2024-10-15:
thanks for that information, it looks like an USB 2.0 device, only USB-LS (1.0) devices are supported

    bcdUSB               2.00

USB-LS HID devices are typically vintage stuff from the 90's, anything newer than that has probably moved away from USB 1.0 and isn't supported

tobozo, 2024-10-15:
indeed it is misleading

https://askubuntu.com/a/395754

this kind of cheap keyboard is quite recent and should work, I used one to develop this library

[image]

 they're called "mini keyboard" and are made in China