tobozo/ESP32-USB-Soft-Host #26: how import  freertos ?

doudousha, 2022-10-22:
d:\arduino\libraries\ESP32-USB-Soft-Host\src/usb_host.h:4:10: fatal error: freertos/FreeRTOS.h: No such file or directory
    4 | #include "freertos/FreeRTOS.h"

tobozo, 2022-10-22:
bad platformio cache maybe ?

ESP32-C3 has no history with this error message so it's more likely a problem with the environment,

With platformio, purging  the `%userprofile%/.platformio` folder can solve some issues.

With Arduino IDE reinstalling the esp32 board package to the latest version (2.0.5) can also improve things.

doudousha, 2022-10-23:
[image]

tobozo, 2022-10-23:
From the file name in your tab it looks like you're using a partial copy of the example, but that example is composed of two files:

[image]

Also arduino IDE claims it's library version 0.1.2 but the error message you have indicates it's neither 0.1.1 nor 0.1.2, did you install it from the library manager?

With ESP32-C3 you should only see this warning message during compilation: 

```log
#pragma message "memory protection features disabled, templated asm nop() will be used"
```
I'm not sure where you got this `Error: memory prot must be disabled` message from, but it's not part of this repository, 
The full compilation log could provide more clues about that inconsistency.

If it can be of any help, these are the settings I've used to test the compilation:

[image]

```log
Sketch uses 266144 bytes (20%) of program storage space. Maximum is 1310720 bytes.
Global variables use 15556 bytes (4%) of dynamic memory, leaving 312124 bytes for local variables. Maximum is 327680 bytes.
```

doudousha, 2022-10-24:
thank you man,
I don't know where to set in arduino

sdima1357/esp32_usb_soft_host  
only for esp32c3: Component config-> ESP System Setting -> Memory protection-> Disable.

usb_test/main/usb_host.c 
line 244  : memory prot must be disabled

[image]

tobozo, 2022-10-24:
the release is now propagated in the [library registry](https://www.arduinolibraries.info/libraries/esp32-usb-soft-host) and consecutively available in the Arduino Library Manager

please let me know if this solved the issue for you

closed 2023-04-04 by tobozo