tobozo/ESP32-USB-Soft-Host #29: loling d32 pro, can't get two keyboards or a mouse detected

gdanov, 2023-01-14:
using the example as-is, here is the output

```
22:57:08.752 -> USB Soft Host Test for LoLin D32 Pro
22:57:08.752 -> TIMER_BASE_CLK: 80000000, TIMER_DIVIDER:2, TIMER_SCALE: 40000000
22:57:08.752 -> Attaching onDetectCB
22:57:08.752 -> Attaching onDataCB
22:57:08.752 -> Attaching onUSBMessageDecode
22:57:08.752 -> Attaching onTickCB
22:57:08.752 -> Creating message queue
22:57:08.752 -> Setting USB Delay
22:57:08.752 -> Setting delay to 4 ticks
22:57:08.752 -> Free iram before alloc: 196532
22:57:08.752 -> Malloc MAX_DELAY_CODE_SIZE=640
22:57:08.752 -> Making xtensa Opcodes for ESP32/ESP32S2
22:57:08.752 -> moving opecodes to executable memory segment
22:57:08.752 -> Free iram after alloc and before realloc: 196532
22:57:08.789 -> Free iram after realloc to cap_exec: 195876
22:57:08.789 -> Setting up pins
22:57:08.789 -> USB#0 (pins 22 23) is OK!
22:57:08.789 -> READ_BOTH_PINS = 0300
22:57:08.789 -> READ_BOTH_PINS = 0000
22:57:08.789 -> READ_BOTH_PINS = 0000
22:57:08.789 -> READ_BOTH_PINS = 0000
22:57:08.789 -> cpu freq = 240 MHz
22:57:08.789 -> TIME_MULT = 51 
22:57:08.789 -> D=   0 Setting delay to 0 ticks
22:57:08.789 -> Free iram before alloc: 195876
22:57:08.789 -> Realloc MAX_DELAY_CODE_SIZE=640
22:57:08.789 -> Making xtensa Opcodes for ESP32/ESP32S2
22:57:08.789 -> moving opecodes to executable memory segment
22:57:08.824 -> Free iram after alloc and before realloc: 196532
22:57:08.824 -> Free iram after realloc to cap_exec: 195876
22:57:08.824 -> 4800 bits in 1297.000000 uSec 3.700848 MHz  6 ticks in 1.621250 uS
22:57:08.824 -> D= 127 Setting delay to 127 ticks
22:57:08.824 -> Free iram before alloc: 195876
22:57:08.824 -> Realloc MAX_DELAY_CODE_SIZE=640
22:57:08.824 -> Making xtensa Opcodes for ESP32/ESP32S2
22:57:08.824 -> moving opecodes to executable memory segment
22:57:08.824 -> Free iram after alloc and before realloc: 196532
22:57:08.862 -> Free iram after realloc to cap_exec: 195876
22:57:08.862 -> 4800 bits in 3811.000000 uSec 1.259512 MHz  6 ticks in 4.763750 uS
22:57:08.862 -> D=  63 Setting delay to 63 ticks
22:57:08.862 -> Free iram before alloc: 195876
22:57:08.862 -> Realloc MAX_DELAY_CODE_SIZE=640
22:57:08.862 -> Making xtensa Opcodes for ESP32/ESP32S2
22:57:08.862 -> moving opecodes to executable memory segment
22:57:08.862 -> Free iram after alloc and before realloc: 196532
22:57:08.862 -> Free iram after realloc to cap_exec: 195876
22:57:08.862 -> 4800 bits in 2524.000000 uSec 1.901743 MHz  6 ticks in 3.155000 uS
22:57:08.900 -> D=  95 Setting delay to 95 ticks
22:57:08.900 -> Free iram before alloc: 195876
22:57:08.900 -> Realloc MAX_DELAY_CODE_SIZE=640
22:57:08.900 -> Making xtensa Opcodes for ESP32/ESP32S2
22:57:08.900 -> moving opecodes to executable memory segment
22:57:08.900 -> Free iram after alloc and before realloc: 196532
22:57:08.900 -> Free iram after realloc to cap_exec: 195876
22:57:08.900 -> 4800 bits in 3172.000000 uSec 1.513241 MHz  6 ticks in 3.965000 uS
22:57:08.900 -> D= 111 Setting delay to 111 ticks
22:57:08.900 -> Free iram before alloc: 195876
22:57:08.934 -> Realloc MAX_DELAY_CODE_SIZE=640
22:57:08.934 -> Making xtensa Opcodes for ESP32/ESP32S2
22:57:08.934 -> moving opecodes to executable memory segment
22:57:08.934 -> Free iram after alloc and before realloc: 196532
22:57:08.934 -> Free iram after realloc to cap_exec: 195876
22:57:08.934 -> 4800 bits in 3492.000000 uSec 1.374570 MHz  6 ticks in 4.365000 uS
22:57:08.934 -> D= 103 Setting delay to 103 ticks
22:57:08.934 -> Free iram before alloc: 195876
22:57:08.934 -> Realloc MAX_DELAY_CODE_SIZE=640
22:57:08.975 -> Making xtensa Opcodes for ESP32/ESP32S2
22:57:08.975 -> moving opecodes to executable memory segment
22:57:08.975 -> Free iram after alloc and before realloc: 196532
22:57:08.975 -> Free iram after realloc to cap_exec: 195876
22:57:08.975 -> 4800 bits in 3331.000000 uSec 1.441009 MHz  6 ticks in 4.163750 uS
22:57:08.975 -> D=  99 Setting delay to 99 ticks
22:57:08.975 -> Free iram before alloc: 195876
22:57:08.975 -> Realloc MAX_DELAY_CODE_SIZE=640
22:57:08.975 -> Making xtensa Opcodes for ESP32/ESP32S2
22:57:08.975 -> moving opecodes to executable memory segment
22:57:08.975 -> Free iram after alloc and before realloc: 196532
22:57:09.034 -> Free iram after realloc to cap_exec: 195876
22:57:09.034 -> 4800 bits in 3259.000000 uSec 1.472844 MHz  6 ticks in 4.073750 uS
22:57:09.034 -> D=  97 Setting delay to 97 ticks
22:57:09.034 -> Free iram before alloc: 195876
22:57:09.034 -> Realloc MAX_DELAY_CODE_SIZE=640
22:57:09.034 -> Making xtensa Opcodes for ESP32/ESP32S2
22:57:09.034 -> moving opecodes to executable memory segment
22:57:09.034 -> Free iram after alloc and before realloc: 196532
22:57:09.034 -> Free iram after realloc to cap_exec: 195876
22:57:09.034 -> 4800 bits in 3212.000000 uSec 1.494396 MHz  6 ticks in 4.015000 uS
22:57:09.034 -> D=  96 Setting delay to 96 ticks
22:57:09.034 -> Free iram before alloc: 195876
22:57:09.034 -> Realloc MAX_DELAY_CODE_SIZE=640
22:57:09.034 -> Making xtensa Opcodes for ESP32/ESP32S2
22:57:09.034 -> moving opecodes to executable memory segment
22:57:09.034 -> Free iram after alloc and before realloc: 196532
22:57:09.075 -> Free iram after realloc to cap_exec: 195876
22:57:09.075 -> 4800 bits in 3192.000000 uSec 1.503759 MHz  6 ticks in 3.990000 uS
22:57:09.075 -> D=  96 Setting delay to 96 ticks
22:57:09.075 -> Free iram before alloc: 195876
22:57:09.075 -> Realloc MAX_DELAY_CODE_SIZE=640
22:57:09.075 -> Making xtensa Opcodes for ESP32/ESP32S2
22:57:09.075 -> moving opecodes to executable memory segment
22:57:09.075 -> Free iram after alloc and before realloc: 196532
22:57:09.075 -> Free iram after realloc to cap_exec: 195876
22:57:09.075 -> 4800 bits in 3191.000000 uSec 1.504231 MHz  6 ticks in 3.988750 uS
22:57:09.119 -> Setting delay to 98 ticks
22:57:09.119 -> Free iram before alloc: 195876
22:57:09.119 -> Realloc MAX_DELAY_CODE_SIZE=640
22:57:09.119 -> Making xtensa Opcodes for ESP32/ESP32S2
22:57:09.119 -> moving opecodes to executable memory segment
22:57:09.119 -> Free iram after alloc and before realloc: 196532
22:57:09.119 -> Free iram after realloc to cap_exec: 195876
22:57:09.119 -> TRANSMIT_TIME_DELAY = 98 time = 3.990000 error = -0.250000% 
22:57:09.119 -> Diff not in [1-7] range: 0
22:57:09.119 -> USB#1 is disabled by user configuration
22:57:09.119 -> Diff not in [1-7] range: 0
22:57:09.119 -> USB#2 is disabled by user configuration
22:57:09.156 -> Diff not in [1-7] range: 0
22:57:09.156 -> USB#3 is disabled by user configuration
22:57:09.156 -> Seleting SCL (blink) Pin #5
22:57:09.156 -> Creating timer config
22:57:09.156 -> Init timer
22:57:09.156 -> Enable interrupt
22:57:09.156 -> Register ISR
22:57:09.156 -> Start timer
22:57:09.156 -> Attaching Timer Task
22:57:09.156 -> USB Soft Host Group timer task is now running on core #1 with priority 5

```

no level shifters, directly to the pins.

tobozo, 2023-01-14:
hi

Have you tried other pins ? I had a successful results with SCL/SDA on a D32 Pro

```
#define DP_P0  SCL
#define DM_P0  SDA
```

gdanov, 2023-01-14:
yes, I started with them.
Also tried S2, no luck too. Keyboard power led lights on, so at least that's right.

gdanov, 2023-01-14:
increased the log level, now I get
```
3:46:38.188 -> USB0: Ack = 0 Nack = 220 01 pcurrent->cb_Cmd = 2  state = 3 epCount = 0 --0001 00ff 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 
23:46:46.197 -> USB0: Ack = 0 Nack = 444 01 pcurrent->cb_Cmd = 2  state = 3 epCount = 0 --0001 00ff 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 
23:46:54.219 -> USB0: Ack = 0 Nack = 668 01 pcurrent->cb_Cmd = 2  state = 3 epCount = 0 --0001 00ff 00ff 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 
23:47:02.224 -> USB0: Ack = 0 Nack = 824 00 pcurrent->cb_Cmd = 0  state = 0 epCount = 0 --0002 00ff 00ff 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 
23:47:10.222 -> USB0: Ack = 0 Nack = 824 02 pcurrent->cb_Cmd = 0  state = 0 epCount = 0 --0002 00ff 00ff 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 
23:47:18.222 -> USB0: Ack = 0 Nack = 824 02 pcurrent->cb_Cmd = 0  state = 0 epCount = 0 --0002 00ff 00ff 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 
```
depending how I connect the signal wires (btw which pin goes to D+/-?)
no message that usb was detected

gdanov, 2023-01-14:
with level shifter connected like here https://cdn.sparkfun.com/assets/learn_tutorials/2/3/8/5/TXS01018E-Assembly2.jpg I see the led blink rapidly, but still Ack = 0

tobozo, 2023-01-15:
**Reminder: only USB 1.0 devices will work with this library.**

I assume you read the documentation and have already verified this important detail, if the keyboard and mice you try aren't in the supported list and their status is unknow, please provide their respective USB PIV/VID.

> with level shifter connected

Level shifter will only make things worse with an esp32, it isn't needed.

Looking at the esp32 pinout, the pins you know for sure aren't pulled up/down are:

2, 4, 13, 15, 25, 26

So if SDA/SCL didn't work, you can still try with {2.4}, {4,2}, {13,15}, {15,13}, {25,26} and {26,25}.

It's worthless to try other combinations, the bitwise hack allowing USB1.0 emulation on ESP32 creates a situation where the "distance" between D+ and D- can't exceed 8 (i.e. using pins {4,13} will never work because 13 - 4 = 9).

[image]

gdanov, 2023-01-16:
Thanks. I wasn't aware it's strictly usb 1.0 only. I guess this is the problem.

closed 2023-01-16 by gdanov