tobozo/ESP32-USB-Soft-Host #31: Merging forks

tobozo, 2023-03-04:
@suarezvictor [your fork](https://github.com/tobozo/ESP32-USB-Soft-Host/compare/main...suarezvictor:ESP32-USB-Soft-Host:main) is very impressive and I wish I found it earlier.

Before starting to integrate your changes, I need to establish an exhaustive board list to handle to the workflow matrix that will run the build tests. Did I miss any device?

- ESP32-Wroom
- ESP32-Wrover
- ESP32-C3
- ESP32-S2
- ESP32-S3
- Teensy

suarezvictor, 2023-03-04:
Thanks! I'd add FPGA soft-cores
By the way, development continued here
https://github.com/suarezvictor/litex_imgui_usb_demo/tree/main/Libs/usbhost

We should continue merging from there and maybe delete this old repo

suarezvictor, 2023-03-04:
I'd really like support for FS devices at  12mbps, and that way test a game controller

tobozo, 2023-03-04:
> I'd really like support for FS devices at 12mbps
12mbps sound way over limit with the USB 1.0 specifications of this USB driver, unless you're talking of using tinyusb?

> Do you know of a good library that does good packet decoding of various types of hardware?
The [USB_Host_Shield](https://github.com/felis/USB_Host_Shield_2.0) library is a very good source of inspiration.

>  I wasn't able to find even a mouse handler that supports both 8-bit protocol and 12-bit protocol
Neither did I when I played with the ESP32-WUD, so I made a [tinyusb descriptor for absolute mouse](https://github.com/tobozo/WUD-Ducky/blob/arduino-esp32-2.0.3-RC1/USBDevices/USBAbsMouse.h) that uses signed 16bits values.

suarezvictor, 2023-03-04:
I'm talking about pushing this exact core to 12mbps, the teensy runs at 600MHz...
Great to know about absolute mouse! they may use it with touchscreens, I have one and didn't figure out how it works, they may use absolute coordinates
regardind the host shield library I'll take a look

what about keeping a C interface and a C++ wrapper for those that require it? C++ in FPGA environments is a bit harder

tobozo, 2023-03-04:
have you seen the this [teensy library](https://github.com/PaulStoffregen/USBHost_t36)? it seems to handle mass storage, filesystem and game controllers altogether

separating C++ wrapper from C interface sounds good but I've never tried FPGA devices before, do I need additional hardware such as an oscilloscope or other special tools to play with that?

suarezvictor, 2023-03-04:
yes, the teeny library may be useful
I've seen a more complete one but can't remember it now
btw such a library doesn't support 12-bit mouse protocol (all my mouses are like that so it may be the norm), see https://github.com/PaulStoffregen/USBHost_t36/blob/master/mouse.cpp#L63

In regards to the C interface, i think that's mandatory. you can do C++ in the FPGA but it's not as usual (indeed I implemented a complete UI with mouse/keyboard support in C++ and using C++ libraries)

go get a FPGA! you need nothing for simple to intermediate things
indeed you can run simulated designs with verilator on your PC at useable speeds (i.e. run a linux console with a custom CPU done in the FPGA)

tobozo, 2023-03-04:
https://github.com/cociorbaandrei/LOGITacker_nrf52832/blob/master/logitacker/logitacker_usb.h this one has a 12bit report descriptor.

suarezvictor, 2023-03-04:
a good driver should detect the protocol, like windows and Linux do
I've seen no embedded development with such a capability

suarezvictor, 2023-03-04:
are we talking about doing a USB host right? so the host hast to adapt to the device capabilities

suarezvictor, 2023-03-04:
we need to split the development in a module that communicates with USB devices, an other that does the decoding and interpretation of packets, the latest one should be really general... just one implementation for all platforms

tobozo, 2023-03-05:
Tried for a dozen hours but I can't get my head around those tinyusb changes, whatever I try doesn't compile with ESP32. I guess it's enough research for a sunday :-)

closed 2025-05-08 by tobozo