tobozo/ESP32-USB-Soft-Host #32: Error compiling example for ESP32-S2

zeezooX, 2023-04-04:
I cannot compile the USB_Test example for my ESP32-S2 board.
I have the latest versions for both the library and the Arduino core.

<details>
<summary>Full error messages</summary>

```
Arduino: 1.8.16 (Windows 10), Board: "ESP32S2 Dev Module, Disabled, Disabled, Disabled, Disabled, UART0, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi), QIO, 80MHz, 4MB (32Mb), 921600, None, Disabled"

In file included from D:\Documents\Arduino\libraries\ESP32-USB-Soft-Host-main\src/ESP32-USB-Soft-Host.h:5,

                 from D:\Documents\Arduino\libraries\ESP32-USB-Soft-Host-main\examples\USB_Test\USB_Test.ino:3:

D:\Documents\Arduino\libraries\ESP32-USB-Soft-Host-main\src/ESP32-USBSoftHost.hpp:18:19: note: #pragma message: memory protection features disabled, templated asm nop() will be used

   #pragma message "memory protection features disabled, templated asm nop() will be used"

                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

In file included from D:\Documents\Arduino\libraries\ESP32-USB-Soft-Host-main\src\ESP32-USBSoftHost.cpp:2:

d:\documents\arduino\libraries\esp32-usb-soft-host-main\src\esp32-usbsofthost.hpp:18:19: note: #pragma message: memory protection features disabled, templated asm nop() will be used

   #pragma message "memory protection features disabled, templated asm nop() will be used"

                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

c:/users/ziyad/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: libraries\ESP32-USB-Soft-Host-main\ESP32-USBSoftHost.cpp.o: in function `op0x00()':

d:\documents\arduino\libraries\esp32-usb-soft-host-main\src/nops.hpp:27: multiple definition of `op0x00()'; sketch\USB_Test.ino.cpp.o:D:\Documents\Arduino\libraries\ESP32-USB-Soft-Host-main\src/nops.hpp:27: first defined here

c:/users/ziyad/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: libraries\ESP32-USB-Soft-Host-main\ESP32-USBSoftHost.cpp.o: in function `op0x01()':

d:\documents\arduino\libraries\esp32-usb-soft-host-main\src/nops.hpp:27: multiple definition of `op0x01()'; sketch\USB_Test.ino.cpp.o:D:\Documents\Arduino\libraries\ESP32-USB-Soft-Host-main\src/nops.hpp:27: first defined here

c:/users/ziyad/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: libraries\ESP32-USB-Soft-Host-main\ESP32-USBSoftHost.cpp.o: in function `op0x02()':

d:\documents\arduino\libraries\esp32-usb-soft-host-main\src/nops.hpp:27: multiple definition of `op0x02()'; sketch\USB_Test.ino.cpp.o:D:\Documents\Arduino\libraries\ESP32-USB-Soft-Host-main\src/nops.hpp:27: first defined here

c:/users/ziyad/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: libraries\ESP32-USB-Soft-Host-main\ESP32-USBSoftHost.cpp.o: in function `op0x03()':

d:\documents\arduino\libraries\esp32-usb-soft-host-main\src/nops.hpp:27: multiple definition of `op0x03()'; sketch\USB_Test.ino.cpp.o:D:\Documents\Arduino\libraries\ESP32-USB-Soft-Host-main\src/nops.hpp:27: first defined here

c:/users/ziyad/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: libraries\ESP32-USB-Soft-Host-main\ESP32-USBSoftHost.cpp.o: in function `op0x04()':

d:\documents\arduino\libraries\esp32-usb-soft-host-main\src/nops.hpp:27: multiple definition of `op0x04()'; sketch\USB_Test.ino.cpp.o:D:\Documents\Arduino\libraries\ESP32-USB-Soft-Host-main\src/nops.hpp:27: first defined here

c:/users/ziyad/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: libraries\ESP32-USB-Soft-Host-main\ESP32-USBSoftHost.cpp.o: in function `op0x05()':

d:\documents\arduino\libraries\esp32-usb-soft-host-main\src/nops.hpp:27: multiple definition of `op0x05()'; sketch\USB_Test.ino.cpp.o:D:\Documents\Arduino\libraries\ESP32-USB-Soft-Host-main\src/nops.hpp:27: first defined here

c:/users/ziyad/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: libraries\ESP32-USB-Soft-Host-main\ESP32-USBSoftHost.cpp.o: in function `op0x06()':

d:\documents\arduino\libraries\esp32-usb-soft-host-main\src/nops.hpp:27: multiple definition of `op0x06()'; sketch\USB_Test.ino.cpp.o:D:\Documents\Arduino\libraries\ESP32-USB-Soft-Host-main\src/nops.hpp:27: first defined here

c:/users/ziyad/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: libraries\ESP32-USB-Soft-Host-main\ESP32-USBSoftHost.cpp.o: in function `op0x07()':

d:\documents\arduino\libraries\esp32-usb-soft-host-main\src/nops.hpp:27: multiple definition of `op0x07()'; sketch\USB_Test.ino.cpp.o:D:\Documents\Arduino\libraries\ESP32-USB-Soft-Host-main\src/nops.hpp:27: first defined here

c:/users/ziyad/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: libraries\ESP32-USB-Soft-Host-main\ESP32-USBSoftHost.cpp.o: in function `op0xd8()':

d:\documents\arduino\libraries\esp32-usb-soft-host-main\src/nops.hpp:40: multiple definition of `op0xd8()'; sketch\USB_Test.ino.cpp.o:D:\Documents\Arduino\libraries\ESP32-USB-Soft-Host-main\src/nops.hpp:40: first defined here

c:/users/ziyad/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: libraries\ESP32-USB-Soft-Host-main\ESP32-USBSoftHost.cpp.o: in function `op0xde()':

d:\documents\arduino\libraries\esp32-usb-soft-host-main\src/nops.hpp:40: multiple definition of `op0xde()'; sketch\USB_Test.ino.cpp.o:D:\Documents\Arduino\libraries\ESP32-USB-Soft-Host-main\src/nops.hpp:40: first defined here

c:/users/ziyad/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: libraries\ESP32-USB-Soft-Host-main\ESP32-USBSoftHost.cpp.o: in function `op0xdf()':

d:\documents\arduino\libraries\esp32-usb-soft-host-main\src/nops.hpp:40: multiple definition of `op0xdf()'; sketch\USB_Test.ino.cpp.o:D:\Documents\Arduino\libraries\ESP32-USB-Soft-Host-main\src/nops.hpp:40: first defined here
[... 130 lines cut ...]

d:\documents\arduino\libraries\esp32-usb-soft-host-main\src/nops.hpp:42: multiple definition of `op0xfa()'; sketch\USB_Test.ino.cpp.o:D:\Documents\Arduino\libraries\ESP32-USB-Soft-Host-main\src/nops.hpp:42: first defined here

c:/users/ziyad/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: libraries\ESP32-USB-Soft-Host-main\ESP32-USBSoftHost.cpp.o: in function `op0xfb()':

d:\documents\arduino\libraries\esp32-usb-soft-host-main\src/nops.hpp:42: multiple definition of `op0xfb()'; sketch\USB_Test.ino.cpp.o:D:\Documents\Arduino\libraries\ESP32-USB-Soft-Host-main\src/nops.hpp:42: first defined here

c:/users/ziyad/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: libraries\ESP32-USB-Soft-Host-main\ESP32-USBSoftHost.cpp.o: in function `op0xfc()':

d:\documents\arduino\libraries\esp32-usb-soft-host-main\src/nops.hpp:42: multiple definition of `op0xfc()'; sketch\USB_Test.ino.cpp.o:D:\Documents\Arduino\libraries\ESP32-USB-Soft-Host-main\src/nops.hpp:42: first defined here

c:/users/ziyad/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: libraries\ESP [... cut ...]
c:/users/ziyad/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: libraries\ESP32-USB-Soft-Host-main\ESP32-USBSoftHost.cpp.o:d:\documents\arduino\libraries\esp32-usb-soft-host-main\src/nops.hpp:46: multiple definition of `nops'; sketch\USB_Test.ino.cpp.o:D:\Documents\Arduino\libraries\ESP32-USB-Soft-Host-main\src/nops.hpp:46: first defined here

collect2.exe: error: ld returned 1 exit status

exit status 1

Error compiling for board ESP32S2 Dev Module.

```
</details>

tobozo, 2023-04-04:
hi, thanks for your feedback :+1: 

I have pushed a fix on the master branch of this repository to address this issue, please let me know if this worked for you.

closed 2023-04-04 by tobozo

zeezooX, 2023-04-04:
The fix worked. I can now run the library with no issues. Thank you for the fast response!

commit 8c5e959a (2023-04-04): Moved nops.hpp inclusion to cpp file, fixes #32