tobozo/ESP32-USB-Soft-Host #34: ADB device host emulation ? (single purpose of setting adb tcpip enable)

shodanx2, 2023-04-26:
Hello
I have a very pointed use case for this that might be useful to really a LOT of people.

And it is the ESP32 behaving as an ADB host to execute only one specfic command 

"adb tcpip 5555"

You see, most modern phones are bootloader lock and unrootable. 
It is impossible to execute any code as root unless you are the manufacturer, network provider or google

This is fine for most people but there is one thing that really unlocks a lot of capability from an android phone and it is enable tcpip network function of the adb client inside the phone.

This is a root function, however, if usb debug mode is enabled (unprivileged user can do it) then it is possible for a adb host device to tell the adb client over usb to enable network capability with the command "adb tcpip 5555" with that number being the port it will listen to for commands

The major problem is that this network connectivity will shut itself off on every reboot.
Currently this requires the user taking the phone back to a real computer running windows or linux to re-enable network connectivity.

So, I would like to inquire, what would it take for the ESP32 to perform this adb host task using software host USB ?

My understanding of the ADB is that is similar to a usb serial port but can be a lot faster.
Upon connection, there is a key exchange, the host provides a key fingerprint and the android client devices shows the user the choice to accept this fingerprint

Then the host would sent the command "adb tcpip 5555" (or another port number) and that would be it.

I can sniff this exchange with my oscilloscope to help developpement of this capability

This would allow anyone with a prepared ESP32 to go to a phone, plug it in the charge port and re-enable network functionality

Example use case
I use two dozen phones as cameras, but unfortunately, the camera will sometime crash when used in viewfinder mode for too long. 
The cameras are rigged, pointed to specific locations and I would really prefer not to touch them once they are setup. However sometimes I have to reset them.  With an esp32, I could leave it connected at all time (currently, it's just a charge cable) and it would continually query the device (is your port 5555 responding to connections) and if not it would do the sequence of enabling network connectivity

I need this remote connectivity to stop and start all those camera's recording remotely as well as swap camera settings (pushing xml config files)

I believe that many people are in my situation as well. Perhaps it would work to create a payment pool  for this feature, I certainly would put my 100$ on that !

tobozo, 2023-04-26:
hi,

ESP32-USB-Soft-Host Library only handles USB 1.0 HID devices so it's out of scope by USB version and by device class.

However, the idea is still good and may even be of interest for Android Automotive OS, and you can probably achieve the same goal with a portion of that $100 bugdet (e.g. use an orange pi zero).

shodanx2, 2023-04-26:
Yes, my current plan is to use a Radza Zero for this purpose.

However I would much rather buy 20 ESP32 than Radza Zero ;)

I suspect that ADB can work in fallback usb 1.0 mode

Do you know of a way to force a usb connection to negotiate down to 1.0 like that ? 

I could then capture a waveform of the exchange and see if it's not really just a usb hid serial device renamed as "adb" ?

tobozo, 2023-04-26:
the host and device need both a companion controller for this fallback to work, but the device must be in otg mode (usb 2.0) to be visible by adb so it's probably a lost cause

maybe use a logic analyzer instead of recording a waveform? bonus: you can record both IN and OUT :wink: 

converting this to a discussion as out of scope for an issue