tobozo/ESP32-USB-Soft-Host #40: ESP32-USB-Soft-Host and EEPROM.h and preferences.h

martelnov, 2023-08-12:
Hello, 
I have a problem. It looks like there is a problem working ESP32-USB-Soft-Host  with eeprom.h and preferences.h
A crash occurs when trying to write to flash.

Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed). 

Core  1 register dump:
PC      : 0x400d3d08  PS      : 0x00060035  A0      : 0x80081262  A1      : 0x3ffbf3cc  
A2      : 0x3ffc342c  A3      : 0x00000000  A4      : 0x3ffc3428  A5      : 0x3ffc6670  
A6      : 0x000095c0  A7      : 0x3ffbb648  A8      : 0x800813dc  A9      : 0x00000000  
A10     : 0x00000010  A11     : 0x00000011  A12     : 0x3ffc6670  A13     : 0x3ffc6630  
A14     : 0x3ffbf5d8  A15     : 0x3ffc668c  SAR     : 0x00000020  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x40086618  LEND    : 0x40086623  LCOUNT  : 0x00000000  

Backtrace: 0x400d3d05:0x3ffbf3cc |<-CORRUPTED

ELF file SHA256: 8e021e340189b662

Rebooting...

tobozo, 2023-08-12:
hi,

> Cache disabled but cached memory region accessed

please specify the espressif core version used with your sketch, there's a chance it's too old and needs to be updated to an earlier version.

martelnov, 2023-08-12:
Arduino-ESP32 v 2.0.11

tobozo, 2023-08-12:
see [this post](https://github.com/tobozo/ESP32-USB-Soft-Host/issues/14) where an untested solution is suggested, using this code in your setup, before `USH.init()`:

```
USH.setISRAllocFlag(0)
```

if it doesn't solve the issue, then I'll need much more information e.g. esp32 type, library version, operating system,a code example that reproduces this error, and any other detail you may find relevant.

martelnov, 2023-08-12:
Compilation error: 'class USB_SOFT_HOST' has no member named 'setISRAllocFlag'

tobozo, 2023-08-12:
checkout the master

martelnov, 2023-08-13:
BINGO!
I downloaded version 1.0.4 and it runs fine.

 

Thank you very much.

closed 2025-05-08 by tobozo