tobozo/ESP32-USB-Soft-Host #49: Are Wemos Lolin ESP32-S3 with integrated USB-OTG compatible ?

M3te0r, 2024-02-27:
Hi,
I'm a fairly beginner in the esp scene but I have a project to use a old Razer Nostromo Keypad as a "programmable" macro keypad with HA

But first I want to try with a dumb spare keyboard I have => https://www.wemos.cc/en/latest/s3/s3.html

I have some Wemos LOLIN ESP32-S3 (+mini) at hand and wanted to use the integrated USB-OTG port to plug my keyboard in it...but it doesn't seem to work

From the wemos datasheet D+/D- are on GPIO19/20 but they do not appear on the board itself, are the 19/20 pin internally linked to USB-OTG...idk

Or do I have to use an USB breakout ? But what about the 19/20 pins for D+/D- then ?

I'm new to this so if someone could help me understand what's going on...I've seen other devkit with 19/20 pin + jumper on the back with USB-OTG label like this one : https://fr.aliexpress.com/item/1005004637451669.html?spm=a2g0o.productlist.main.7.1bb4lrB5lrB5VG&algo_pvid=72f72592-110c-4d9f-9796-a3915fba881f&algo_exp_id=72f72592-110c-4d9f-9796-a3915fba881f-3&pdp_npi=4%40dis%21EUR%213.50%213.50%21%21%213.71%213.71%21%402103847817090586028926195ed234%2112000036725496197%21sea%21FR%214009116193%21&curPageLogUid=sZIBwRg1ozyj&utparam-url=scene%3Asearch%7Cquery_from%3A

tobozo, 2024-02-27:
hi, 

may I suggest you try the [USB_Example](https://github.com/tobozo/ESP32-USB-Soft-Host/blob/main/examples/USB_Test/USB_Test.ino) with one of [those](https://fr.aliexpress.com/item/1005005916687413.html) connected the pins 15/16 just to see if your dumb spare keyboard and the Razer Nostromo Keypad get detected.

I couldn't find a lot of information on the Razer Nostromo Keypad, its compatibility with USB 1.0 should be confirmed before thinking about using ESP32-USB-Soft-Host.

tobozo, 2025-05-08:
closing this as inactive, feel free to reopen if you have new information

closed 2025-05-08 by tobozo

9dl, 2025-07-11:
I tried this but without any success sadly, if anyone could help me would be great, i always get "USB#0 disconnected" when disconnecting the USB but nothing else no plugged in or keystrokes logging. I tried soldering the USB OTG Pins at first (19/20) did not work, same with the 15/16 pins.