tobozo/ESP32-USB-Soft-Host #50: SNES Controller not detected

emanuelelaface, 2024-04-27:
This is a bit weird issue.
I spent some time developing a USB to DB9 (Commodore 64) controller using a SNES game pad.
Everything worked well and because of the 2 Commodore 64 ports I decided to buy a second SNES controller, but this second is not even detected.
Of course I thought that it could have been broken, but tested on a Mac it works perfectly. So I decided to open and inside the two controllers seems quite different. In particular the main difference that I can see is that one controller seems to have a crystal and the other not, but I can't see the chip used because it is under a black cap.

I am a bit stuck here because no message, appears on the console so it seems that the device is not detected in any way.
Is there something that I can do to debug this issue?

tobozo, 2024-04-27:
hi,

either one of the two controllers does not support USB-LS (may be limited to USB 2.0) or it's using a bad crystal and the esp32 can't compensate

comparing the details of the two controllers will tell about their capabilities; you'll need a tool like `lsusb`, on MacOS it is part of a packaged named `usbutils`

lsusb can be installed using brew:

    $> brew install usbutils

... or using port:

    $> sudo port install usbutils

run it once without arguments to find the `VendorID:ProductID` of the gamepads in the devices list:

    $> lsusb

    (...)
    Bus 00x Device 00x: ID 0123:4567 Your Gamepad Controller 
    Bus 00x Device 00x: ID 89ab:cdef Your Other Gamepad Controller 
    (...)

then run lsusb with the VendorID/ProductID of that gamepad (e.g. `0123:4567` or `89ab:cdef`)

    $> lsusb -d 0123:4567 -v

copy the results in a text editor, rinse and repeat with the other gamepad, then compare both results

emanuelelaface, 2024-04-27:
I report here the two output of lsusb, but they are very similar:

The diff between working and not working controllers:
```
2c2
< Bus 000 Device 003: ID 0810:e501
---
> Bus 000 Device 003: ID 081f:e401
11,12c11,12
<   idVendor           0x0810
<   idProduct          0xe501
---
>   idVendor           0x081f
>   idProduct          0xe401
45c45
<           wDescriptorLength      89
---
>           wDescriptorLength      98
```

The full output of working controller:
```
Bus 000 Device 003: ID 0810:e501
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0
  bMaxPacketSize0         8
  idVendor           0x0810
  idProduct          0xe501
  bcdDevice            1.06
  iManufacturer           0
  iProduct                2
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 [unknown]
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      0
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode           33 US
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      89
          Report Descriptors:
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10
Device Status:     0x0000
  (Bus Powered)
  ```
  
  The full output of not working controller:
  ```
  Bus 000 Device 003: ID 081f:e401
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0
  bMaxPacketSize0         8
  idVendor           0x081f
  idProduct          0xe401
  bcdDevice            1.06
  iManufacturer           0
  iProduct                2
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 [unknown]
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      0
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode           33 US
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      98
          Report Descriptors:
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10
Device Status:     0x0000
  (Bus Powered)
  ```

tobozo, 2024-04-27:
according to this both controllers are USB 1.0

    bcdUSB               1.00

indeed some USB-LS devices with bad or no crystal can fail to be detected by ESP32-USB-Soft-Host, but this needs to be confirmed:

- slightly changing the voltage can affect the signal, is it powered by 5v exactly or is it using the VCC pin from the ESP? 
- cable length can affect the signal, if possible confirm it also fails with shorter clip wires
- different port can give different results, ESP32-USB-Soft-Host supports up to 4
- further debugging requires an oscilloscope but this is beyond my knowledge and skills

emanuelelaface, 2024-04-27:
Unfortunately none of the first three things made a difference.
I tried both with VCC and +5V, I swapped ports (I enabled just 2 but) and tried with different cables.

I am available also to test with an oscilloscope but I do not know exactly what to test.

Thanks a lot for your help anyway.

tobozo, 2024-05-01:
> I am available also to test with an oscilloscope but I do not know exactly what to test.

you can verify that the bit rate / bit period are within the clock tolerance range (1.50±0.18 Mbit/s) in all 4 situations

if you feel adventurous you can also use [wireshark + logic analyzer](https://re-ws.pl/2022/08/sniffing-usb-traffic-with-dslogic-logic-analyzer-into-pcap-file/) to compare data between working and failing scenarios i.e. to identify where the fail happens (voltage, handshake, setup, data, crc)

emanuelelaface, 2024-05-02:
Thank you, I will check what I can do and see if I understand what is going on.