tobozo/ESP32-USB-Soft-Host #51: Error compiling code on esp32-s3

mttcarbone, 2024-05-29:
Hi, 
I was compiling the code on esp32-s3 OTG and I get these errors
how can they be fixed? 

In file included from /home/mc/Arduino/libraries/ESP32-USB-Soft-Host/src/ESP32-USB-Soft-Host.h:5,
                 from /tmp/.arduinoIDE-unsaved2024428-68974-1kyf6pp.qsvb/sketch_may28a/sketch_may28a.ino:13:
/home/mc/Arduino/libraries/ESP32-USB-Soft-Host/src/ESP32-USBSoftHost.hpp:13:24: error: 'LED_BUILTIN' was not declared in this scope
     #define BLINK_GPIO LED_BUILTIN
                        ^~~~~~~~~~~
/home/mc/Arduino/libraries/ESP32-USB-Soft-Host/src/ESP32-USBSoftHost.hpp:98:41: note: in expansion of macro 'BLINK_GPIO'
     gpio_num_t blink_gpio = (gpio_num_t)BLINK_GPIO;
                                         ^~~~~~~~~~
/home/mc/Arduino/libraries/ESP32-USB-Soft-Host/src/ESP32-USBSoftHost.hpp:13:24: note: suggested alternative: 'LED_NUMLOCK'
     #define BLINK_GPIO LED_BUILTIN
                        ^~~~~~~~~~~
/home/mc/Arduino/libraries/ESP32-USB-Soft-Host/src/ESP32-USBSoftHost.hpp:98:41: note: in expansion of macro 'BLINK_GPIO'
     gpio_num_t blink_gpio = (gpio_num_t)BLINK_GPIO;
                                         ^~~~~~~~~~

tobozo, 2024-05-29:
hi,

what board did you select in the arduino ide tools menu? 

if you know what esp32 pin is connected to the builtin led, you can specify it before including the usb soft host library:

```C
// if the onboard led is on pin 2
#define LED_BUILTIN GPIO_NUM_2 
```

mttcarbone, 2024-05-29:
The board selected on the menu is ESP32-S3-USB-OTG. 
If I go to define `BLINK_GPIO` and `LED_BUILTIN` in the ESP32-S3-USB_Test sketch, the only error that remains is this:

In file included from /home/mattiacarbone/Arduino/libraries/ESP32-USB-Soft-Host/src/ESP32-USB-Soft-Host.h:5,
                 from /tmp/.arduinoIDE-unsaved2024428-68974-1kyf6pp.qsvb/sketch_may28a/sketch_may28a.ino:15:
/home/mattiacarbone/Arduino/libraries/ESP32-USB-Soft-Host/src/ESP32-USBSoftHost.hpp:98:40: error: expected primary-expression before ')' token
     gpio_num_t blink_gpio = (gpio_num_t)BLINK_GPIO;

tobozo, 2024-05-29:
> If I go to define BLINK_GPIO and LED_BUILTIN 

BLINK_GPIO is internal to esp32-usb-soft-host, if you define that one before including the library, don't define LED_BUILTIN

do you have an onboard led btw?

if not you can try this:

```cpp
#define DP_P0  15  // your USB pin for D+ (edit this)
#define DM_P0  16  // your USB pin for D- (edit this)
#define FORCE_TEMPLATED_NOPS // esp32s3 will work better with this flag
#define BLINK_GPIO -1 // -1 = no onboard led
#include <ESP32-USB-Soft-Host.h>

usb_pins_config_t USB_Pins_Config =
{
  DP_P0, DM_P0,
  -1, -1,
  -1, -1,
  -1, -1
};

void setup()
{
  Serial.begin(115200);
  USH.setActivityBlinker(nullptr); // disable activity led blinking
  USH.init(USB_Pins_Config);
}
```

if you still get errors I'll need more info: OS, Arduino IDE version, espressif core version, and the full source of the sketch you're running

mttcarbone, 2024-05-29:
Entering the code you shared returned these errors:

In file included from /home/mattiacarbone/Arduino/libraries/ESP32-USB-Soft-Host/src/ESP32-USBSoftHost.cpp:2:
/home/mattiacarbone/Arduino/libraries/ESP32-USB-Soft-Host/src/./ESP32-USBSoftHost.hpp:13:24: error: 'LED_BUILTIN' was not declared in this scope
     #define BLINK_GPIO LED_BUILTIN
                        ^~~~~~~~~~~
/home/mattiacarbone/Arduino/libraries/ESP32-USB-Soft-Host/src/./ESP32-USBSoftHost.hpp:98:41: note: in expansion of macro 'BLINK_GPIO'
     gpio_num_t blink_gpio = (gpio_num_t)BLINK_GPIO;
                                         ^~~~~~~~~~
/home/mattiacarbone/Arduino/libraries/ESP32-USB-Soft-Host/src/ESP32-USBSoftHost.cpp:5:19: note: #pragma message: memory protection features disabled, templated asm nop() will be used
   #pragma message "memory protection features disabled, templated asm nop() will be used"
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

I am currently testing it on the usb host, with only the sample code provided by your library. I am using Arduino IDE 2.3.2 on Pop_os 22.04 (ubuntu) and in the board manager I have installed esp32 from espressif at version 2.0.11. 
I hope this information is sufficient. 
This is the pinout of the board used by me (ESP32 S3 DevKitC 1 N16R8)
[image]

Thanks for your help

tobozo, 2024-05-29:
thanks, I'll try to reproduce that and think of a way to prevent those errors.

meanwhile you can try to bypass the error by testing the [esp32s3 example sketch](https://github.com/tobozo/ESP32-USB-Soft-Host/tree/main/examples/ESP32-S3-USB_Test) using  `ESP32S3 Dev module` board profile instead of `ESP32-S3-USB-OTG`

mttcarbone, 2024-05-29:
Thank you! 
Selecting the ESP32S3-DEV no longer gives me any errors. 
I ask you, how can I print in the serial of the esp32 what I type with the keyboard? Is there a special function for this? Does it need any special parameters? Reading the documentation, I didn't understand this aspect.

closed 2025-05-08 by tobozo