tobozo/ESP32-USB-Soft-Host #52: this sequence goes on endlessly after the device is connected

ferakond, 2024-06-13:
New device detected on USB#0
desc.bcdUSB             = 0x0100
desc.bDeviceClass       = 0x00
desc.bDeviceSubClass    = 0x00
desc.bDeviceProtocol    = 0x00
desc.bMaxPacketSize0    = 0x08
desc.idVendor           = 0x081f
desc.idProduct          = 0xe401
desc.bcdDevice          = 0x0106
desc.iManufacturer      = 0x00
desc.iProduct           = 0x02
desc.iSerialNumber      = 0x00
desc.bNumConfigurations = 0x01
Config Descriptor #1
  cfg.wLength         = 0x22
  cfg.bNumIntf        = 0x01
  cfg.bCV             = 0x01
  cfg.bMaxPower       = 250
    Interface Descriptor #1
      sIntf.bLength     = 0x09
      sIntf.bType       = 0x04
      sIntf.iNum        = 0x00
      sIntf.iAltString  = 0x00
      sIntf.bEndPoints  = 0x01
      sIntf.iClass      = 0x03
      sIntf.iSub        = 0x00
      sIntf.iProto      = 0x00
      sIntf.iIndex      = 0x00
        HID Device Descriptor #1
          hid.bLength               = 0x09
          hid.bDescriptorType       = 0x21
          hid.bcdHID                = 0x110
          hid.bCountryCode          = 0x21
          hid.bNumDescriptors       = 0x01
          hid.bReportDescriptorType = 0x22
          hid.wItemLengthL          = 0x62
          hid.wItemLengthH          = 0x00
      EndPoint Descriptor #1
        epd.bLength       = 0x07
        epd.bType         = 0x05
        epd.bEPAdd        = 0x81
        epd.bAttr         = 0x03
        epd.wPayLoad      = 0x08
        epd.bInterval     = 0x0a
in: 0x7f 0x7f 0x00 0x80 0x80 0x0f 0x00 0x00 
in: 0x7f 0x7f 0x00 0x80 0x80 0x0f 0x00 0x00 
in: 0x7f 0x7f 0x00 0x80 0x80 0x0f 0x00 0x00 
in: 0x7f 0x7f 0x00 0x80 0x80 0x0f 0x00 0x00 
in: 0x7f 0x7f 0x00 0x80 0x80 0x0f 0x00 0x00 
in: 0x7f 0x7f 0x00 0x80 0x80 0x0f 0x00 0x00

tobozo, 2024-06-14:
from the vendor database:

  idVendor           0x081f 
  idProduct          0xe401 
  Generic Controllers / SNES USB Controller
  MaxPower              500mA

from the init:

  cfg.bMaxPower = 250

could it mean the ESP cannot provide 500mA and the snes gamepad negociates a lower voltage it can't sustain?

ferakond, 2024-06-14:
I'll try another power source. If I understood you correctly, of course. When powered by usb 3.0, it was the same