tobozo/ESP32-USB-Soft-Host #54: VCP - virtual com port, is it possible in theory?

latonita, 2024-10-16:
Hi,

I started investigation on how I can connect USB serial device (9600 bps) no more.
Do you think this is possible with USB soft implementation (I realize it shall be enhanced/implemented).

Thanks

tobozo, 2024-10-16:
Hi,

I don't know, 9600 sounds reasonable but most CDC devices are at least using USB 1.1, which is out of specs for ESP32-USB-Soft-Host, any solution involving an esp32-wroom soft-host driver would have to be very hacky to work.

You'll probably be better off using tinyusb with a raspi pico or an esp32-s2/s3.

Rob58329, 2025-02-26:
Is it possible to connect the "integrated USB Serial/JTAG Controller" USB-port of an ESP32C3 (or even another ESP32S3) (VID=303A, PID=1001) instead of a keyboard, and read the incoming serial-data?

{I note that https://github.com/luc-github/esp32-usb-serial works with an ESP32S3 DevKitC1 in USB-Host mode (host_ftdi, host_cp210x and host_ch34x) allowing the reading of serial-data from an FDTI or CP210x or CH34x USB to TTL Serial UART.)